shimming-toolbox/rf-shimming-7t #31: Fix spinal cord segmentation

jcohenadad, 2024-01-09:
Two subjects failing:

[image]

[image]

jcohenadad, 2024-01-09:
Using `-c t2s` they all look good! ✅ 

[image]

[image]

[image]

[image]

Except sub-04:

[image]

So I can make a close in the code to use `-c t2` only for sub-04

dpapp86, 2024-01-10:
Sorry to reopen.

I am not sure I understand. Are we switching back to sct_deepseg_sc for this project?

Have you checked other cases? In my experience, at least half the rf shim cases require some manual correction, especially lower down (slices 0-10). MONAI mislocalized or even fully missed the sc, see here for sub-05, patspec, slice 12
[image]

jcohenadad, 2024-01-10:
> Are we switching back to sct_deepseg_sc for this project?

Yes we are. 

> Have you checked other cases? In my experience, at least half the rf shim cases require some manual correction, especially lower down (slices 0-10)

Yes, QC report is generated. All slices covering C1-T2 look fine to me with the current code.

jcohenadad, 2024-01-16:
Re-closing as https://github.com/shimming-toolbox/rf-shimming-7t/issues/31#issuecomment-1885746218 is addressed

PR #32 (2024-01-09): Use another segmentation method for sub-04
- Fixes #31
closed 2024-01-16 by jcohenadad