shimming-toolbox/rf-shimming-7t #35: Register the other shim methods to the GRE CoV scan

jcohenadad, 2024-01-10:


dpapp86, 2024-01-11:
Since sub-04 is the one with the worst image quality (chunky lad), I propose using it as a benchmark.

Likewise, patient or volume tend to have the lowest signal intensity/contrast, so they are a good worst case scenario. For sub-04, I am looking at CoV-2-CoV (good sanity check), CP-2-CoV, Volume-2-CoV, and Target-2-CoV. I looked at a one-step coreg using affine, slicereg  and bsplinesyn. And dl for good measure.

Code in question:
``for subject in subjects:
    os.chdir(os.path.join(path_data, subject, "anat"))
    if subject == 'sub-04':
        for shim_mode in shim_modes:
            !sct_register_multimodal -i {subject}_acq-{shim_mode}_T2starw.nii.gz -d {subject}_acq-CoV_T2starw_crop.nii.gz -dseg ../anat/{subject}_acq-CoV_T2starw_crop_seg.nii.gz -param step=1,type=im,algo=slicereg,metric=CC -qc {path_qc}``

**Using affine:**

Quite bad on patient, see jumping SC at cursor:

https://github.com/shimming-toolbox/rf-shimming-7t/assets/80138067/4a926fbc-8089-414b-809c-78c73ed7de98

**Using slicereg:**
Same jumping:

https://github.com/shimming-toolbox/rf-shimming-7t/assets/80138067/2de721f7-ab6e-4235-9f29-00aec31ec3c6

**Using bsplinesyn:**
Very poor

https://github.com/shimming-toolbox/rf-shimming-7t/assets/80138067/913a74bc-1fef-4cd8-96ea-26c2d4d8e956

**Using DL:**
Best one so far

https://github.com/shimming-toolbox/rf-shimming-7t/assets/80138067/5dc7b7eb-99cb-4d42-a630-231eb896cfbf

What is your opinion @jcohenadad ?

jcohenadad, 2024-01-17:
Thank you for these investigations. The DL algo definitely looks promising. 

I've had hard time doing a proper registration on sub-04, see preliminary investigations in https://github.com/shimming-toolbox/rf-shimming-7t/pull/57#issuecomment-1894615051. 

The issue is partly because of https://github.com/shimming-toolbox/rf-shimming-7t/issues/58. 

However, given that we don't _absolutely_ need to report GRE signal across all shim modes (I think there is enough material in the paper so far), I would skip that step for now.

PR #65 (2024-01-17): Register the other shim methods to the GRE CoV scan
Fixes #35
commit 61f67c3f (2024-01-17): Replacing slicereg by deep learning registration

Fixes #35
closed 2024-01-17 by dpapp86