shimming-toolbox/rf-shimming-7t #38: What cervical levels to plot and calculate between?

dpapp86, 2024-01-10:
As of now, B1+ maps are processed and plotted up to the top of C1. 

However, the RF shimming weights are only calculated between C2/C3 and T2/T3. What is the reasoning for including the top to cervical vertebral levels, when we did _not_ shim them?

jcohenadad, 2024-01-10:
> What is the reasoning for including the top to cervical vertebral levels, when we did not shim them?

No good reason. I agree we should perform analysis starting from C2.

dpapp86, 2024-01-21:
Reopening as of [bf4cf2f](https://github.com/shimming-toolbox/rf-shimming-7t/commit/bf4cf2f934e521c7e25fe1e062d43e5175bf433f)

Both the CSF/SC figure and the B1+ mapping code show the resampling being done on the C1-T2 range, as opposed to the C3:T2 range, where the values have been extracted. I guess this results in stretching the plots.

[#](https://github.com/shimming-toolbox/rf-shimming-7t/blob/main/data_processing.ipynb?short_path=e237aea#L352)
and 
[#](https://github.com/shimming-toolbox/rf-shimming-7t/blob/main/data_processing.ipynb?short_path=e237aea#L560)

commit 36bb2d7a (2024-01-16): Drop levels C1 and C2 before performing statistics

Fixes #38
commit febb3c64 (2024-01-24): Sample and display CSF/SC T2starw and B1+ between C3-T2

Fixes #38, Fixes #75
closed 2024-01-24 by jcohenadad