shimming-toolbox/rf-shimming-7t #49: Consider doing the segmentation and vertebral labeling only on the MPRAGE

jcohenadad, 2024-01-16:
Consider doing the segmentation and vertebral labeling only on the MPRAGE, instead of doing it _also_ on the GRE scan. In this scenario, the GRE scan needs to be robustly registered with the T1w scan.

dpapp86, 2024-01-16:
I recall spending time on this issue without a satisfactory solution, at least when the GRE segmentation is not present. There is always a bit too much stretching/shrinking/distortion. I did not try on cropped images, perhaps that helps.

Can update later today, please assign it as a task if you wish for me to do so.

jcohenadad, 2024-01-16:
I found an excellent solution with slicereg and the cord segmentation on the GRE (which is robust): [qc.zip](https://github.com/shimming-toolbox/rf-shimming-7t/files/13952645/qc.zip)

Will implement now and open a PR

closed 2024-01-17 by jcohenadad