shimming-toolbox/rf-shimming-7t #58: Strange deformation within session for `sub-04`

jcohenadad, 2024-01-16:
While working on #57, I noticed strange deformation within a session for `sub-04`:

[image]

I suspect the table was moved significantly between these two acquisitions, resulting in drastically different gradient non-linearity distortions. 

@dpapp86 Do you recall something abnormal for this acquisition?

dpapp86, 2024-01-17:
@jcohenadad Here is what I found:

0, Which two acquisitions are these?

1, The Table Postion DICOM tag is the same for all GRE (and tfl_rfmap and tfl_b1map and MPRAGE) scans
2, The shim settings are all the same across RF shimming approaches
3, The CP mode T2star is the one **WITH** ditrotion correction! Could that explain it? As it stands in our code, **none** of the GRE scans we process have GDC, see #11 
[image]

jcohenadad, 2024-01-17:
> 3, The CP mode T2star is the one WITH ditrotion correction! Could that explain it? As it stands in our code, none of the GRE scans we process have GDC, see https://github.com/shimming-toolbox/rf-shimming-7t/issues/11

Ah! Amazing catch! That must be it. I probably uploaded the wrong image for some reasons. Thank you.

dpapp86, 2024-01-17:
I did a quick eval, every other scan is OK:

[image]

And the same is true for the fmap anat scans

jcohenadad, 2024-01-19:
Fixed in: https://openneuro.org/datasets/ds004906/versions/2.3.0

closed 2024-01-19 by jcohenadad