shimming-toolbox/rf-shimming-7t #71: Process anat/T2starw (GRE) step: manual segmentations not found

dpapp86, 2024-01-21:
Current as of [bf4cf2f](https://github.com/shimming-toolbox/rf-shimming-7t/commit/bf4cf2f934e521c7e25fe1e062d43e5175bf433f)

Used a clean download of the datalad data.

The "Process anat/T2starw (GRE) step" does not find the manual segmentations anymore. I guess this is related to #67 , but at at quick parse I can't find a corresponding upload for this.

The code is looking for
` f"{subject}_acq-CoV_T2starw_label-SC_seg.nii.gz")`
but the data only consists of:
[image]

Is this WAD?

dpapp86, 2024-01-21:
@jcohenadad Ping as promised

jcohenadad, 2024-01-22:
I cannot reproduce what you're experiencing @dpapp86, so I would need more details about what you did exactly (and what you refer to by "manual segmentation": are you referring to the SC? the CSF? both?)

Eg: running a colab notebook based on https://github.com/shimming-toolbox/rf-shimming-7t/commit/bf4cf2f934e521c7e25fe1e062d43e5175bf433f, This is my output for the cell "# Copy CSF masks from the derivatives folder":

[image]

Showing that the CSF seg was found and copied.

jcohenadad, 2024-01-23:
Weirdly, when running the notebook locally, the manual segmentation is _not_ found. Investigating... 🔎

dpapp86, 2024-01-23:
Sorry for the lateish reply, time zone differences.

I always run the notebook locally, because I am too impatient to wait for the SCT install. But I can see that you have managed to reproduce.

jcohenadad, 2024-01-23:
The full derivatives are not present on my local station. Output of datalad code:

<details><summary>Terminal output</summary>

```console
Cloning:   0%|                             | 0.00/2.00 [00:00<?, ? candidates/s]
Enumerating: 0.00 Objects [00:00, ? Objects/s]
                                              
Counting:   0%|                              | 0.00/2.48k [00:00<?, ? Objects/s]
                                                                                
Compressing:   0%|                           | 0.00/1.30k [00:00<?, ? Objects/s]
                                                                                
Receiving:   0%|                             | 0.00/2.48k [00:00<?, ? Objects/s]
Receiving:  46%|█████████▏          | 1.14k/2.48k [00:00<00:00, 10.7k Objects/s]
                                                                                
Resolving:   0%|                                | 0.00/698 [00:00<?, ? Deltas/s]
[INFO   ] Remote origin not usable by git-annex; setting annex-ignore           
[INFO   ] https://github.com/OpenNeuroDatasets/ds004906.git/config download failed: Not Found 
install(ok): /Users/julien/code/rf-shimming-7t/ds004906 (dataset)
Total:   0%|                                    | 0.00/524M [00:00<?, ? Bytes/s]
Get code/bid .. onversion.py:   0%|            | 0.00/3.20k [00:00<?, ? Bytes/s]
Total:   0%|                        | 3.20k/524M [00:00<18:18:04, 7.96k Bytes/s]
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/48.4k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. F_seg.nii.gz:   0%|            | 0.00/56.6k [00:00<?, ? Bytes/s]
Total:   0%|                             | 108k/524M [00:00<51:19, 170k Bytes/s]
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/46.3k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/48.4k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. F_seg.nii.gz:   0%|            | 0.00/56.8k [00:00<?, ? Bytes/s]
Total:   0%|                             | 260k/524M [00:00<28:22, 308k Bytes/s]
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/46.3k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/48.4k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. F_seg.nii.gz:   0%|            | 0.00/54.6k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. C_seg.nii.gz:   0%|            | 0.00/49.7k [00:00<?, ? Bytes/s]
Total:   0%|                             | 459k/524M [00:01<21:19, 410k Bytes/s]
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/46.3k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/48.3k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. w_seg.nii.gz:   0%|            | 0.00/54.2k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. F_seg.nii.gz:   0%|            | 0.00/56.6k [00:00<?, ? Bytes/s]
Total:   0%|                             | 664k/524M [00:01<18:25, 474k Bytes/s]
Get derivati .. C_seg.nii.gz:   0%|            | 0.00/50.7k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/46.3k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/48.3k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. w_seg.nii.gz:   0%|            | 0.00/54.5k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. F_seg.nii.gz:   0%|            | 0.00/56.3k [00:00<?, ? Bytes/s]
Total:   0%|                             | 920k/524M [00:01<16:43, 522k Bytes/s]
Get derivati .. C_seg.nii.gz:   0%|            | 0.00/50.8k [00:00<?, ? Bytes/s]
                                                                                
Get derivati .. _dseg.nii.gz:   0%|            | 0.00/46.3k [00:00<?, ? Bytes/s]
                                                                                
Get sub-01/a .. P_T1w.nii.gz:   0%|            | 0.00/9.45M [00:00<?, ? Bytes/s]
Total:   2%|▌                          | 10.5M/524M [00:02<01:49, 4.68M Bytes/s]
Get sub-01/a .. starw.nii.gz:   0%|            | 0.00/10.3M [00:00<?, ? Bytes/s]
Get sub-01/a .. starw.nii.gz:  32%|▉  | 3.34M/10.3M [00:00<00:00, 33.3M Bytes/s]
Total:   4%|█                          | 20.8M/524M [00:02<01:04, 7.84M Bytes/s]
Get sub-01/a .. V_T1w.nii.gz:   0%|            | 0.00/9.72M [00:00<?, ? Bytes/s]
                                                                                
Get sub-01/a .. starw.nii.gz:   0%|            | 0.00/10.1M [00:00<?, ? Bytes/s]
[... 262 lines cut ...]
                                                                                
[... cut ...]
get(ok): derivatives/labels/sub-03/anat/sub-03_acq-CoV_T1w_label-discs_dseg.nii.gz (file) [from s3-PUBLIC...]
get(ok): derivatives/labels/sub-03/anat/sub-03_acq-CoV_T2starw_label-CSF_seg.nii.gz (file) [from s3-PUBLIC...]
get(ok): derivatives/labels/sub-03/anat/sub-03_acq-CoV_T2starw_label-SC_seg.nii.gz (file) [from s3-PUBLIC...]
  [126 similar messages have been suppressed; disable with datalad.ui.suppress-similar-results=off]
action summary:
  get (ok: 136)
# Define useful variables
```

</details>

We should add a syntax that indicates what version of the dataset is being downloaded.

jcohenadad, 2024-01-23:
Wait-- my bad: segmentations for sub-01 and sub-02 are _not_ uploaded on OpenNeuro, because their automatic output is perfect. However, sub-03, sub-04, sub-05 are, and they are properly identified by the code:

[image]

Please confirm that sub-03, sub-04, sub-05 are found on your end-- if that's the case you can close the issue

jcohenadad, 2024-01-23:
We can add a comment in the cell, saying that the manual segmentation is uploaded for _some_ subjects, and if the code finds it, it uses it.

dpapp86, 2024-01-23:
This is interesting. 

The begigging of the data download cell's output is apparently trying and failing to grab stuff from the Derivaties folder.
[image]

Then, at the end, the following derivatives are OK:
[image]

The rest is all sub-XX, all downloaded well.

dpapp86, 2024-01-23:
Segmentations NOT found for 01 and 02
[image]
[image]

Segmentations ARE found for 03 04 05:
[image]
[image]
[image]

dpapp86, 2024-01-23:
@jcohenadad 

Closing as per https://github.com/shimming-toolbox/rf-shimming-7t/issues/71#issuecomment-1906348807

closed 2024-01-23 by dpapp86