shimming-toolbox/rf-shimming-7t #75: Wrong coverage for signal extraction

jcohenadad, 2024-01-22:
See comment: https://github.com/shimming-toolbox/rf-shimming-7t/pull/69#discussion_r1461112504

As of https://github.com/shimming-toolbox/rf-shimming-7t/pull/69, there is an inconsistency: we sample between C3-T2, but we display between C1-T2.

Option 1: _only_ extract between C3-T2 (and show C3-T2)
- pros: less contamination of smoothing kernel https://github.com/shimming-toolbox/rf-shimming-7t/issues/74
- cons: plot won't show the region where RF shimming was not applied (hence not possible to qualitatively compare). 

Option 2: extract between C1-T2 and show C1-T2, and do the stats only between C3-T2

dpapp86, 2024-01-23:
Option 2 sounds good. Would it be possible (and correct) to smooth C1-C2 and C3-T2 separately, to avoid the leakage mentioned in #69?

We should also draw a vertical line on the figure to separate the shimmed and unshimmed regions

jcohenadad, 2024-01-24:
I'm going for option 1 in the end-- it makes less complicated code

commit febb3c64 (2024-01-24): Sample and display CSF/SC T2starw and B1+ between C3-T2

Fixes #38, Fixes #75
closed 2024-01-24 by jcohenadad