shimming-toolbox/rf-shimming-7t #77: Consider encapsulating SCT inside a Docker image

jcohenadad, 2024-01-23:
Everytime the notebook is ran on Google Colab, SCT is installed, and that takes at least 5min. Would it be possible/easy to instead use a Docker image with SCT on it?

joshuacwnewton, 2024-01-23:
Before talking about specific solutions, I do want to briefly say:

> Fundamentally, this is more of a problem with Google Colab than it is with SCT's installation method. This is because, even if SCT were pip-installable, we would still have to deal with the problem of Google Colab timing out and not preserving environments. See: This popular [How can I prevent Google Colab from disconnecting?](https://stackoverflow.com/questions/57113226/how-can-i-prevent-google-colab-from-disconnecting) question on StackOverflow, for example. Google doesn't seem to have any plans to fix this issue, either: https://github.com/googlecolab/colabtools/issues/3547

As for your suggestion, it sounds like you're essentially suggesting:

- **Without Docker**: Time spent downloading + Time spent installing dependencies into a venv.
- **With Docker**: Time spent downloading (the Docker image) **only**.

In other words, we would try to "cache" the installed SCT environment, thus saving time during setup.

Some thoughts:

- I think that this "caching" idea does make sense, and could save time. Though, it's unclear to me how much time is spent downloading vs. how much time is spent installing, so I'm not sure how much time this could save. (e.g. is 5m -> 2m a worthwhile difference? 5m -> 3m? etc.)
- Using Docker for this idea:
  - Doing some cursory research, Google Colab itself appears to [run inside a Docker container](https://gist.github.com/mwufi/6718b30761cd109f9aff04c5144eb885?permalink_comment_id=4471586#gistcomment-4471586), which makes things a little more complicated (Docker-in-Docker), as compared to running Docker inside a regular Linux VM.
  - Additionally, I worry that the installation of Docker itself may take quite a bit of time, undoing any benefits we would get by creating an SCT Docker image.
- Using `udocker` for this idea:
  - It seems, though, that folks have used [`udocker`](https://github.com/indigo-dc/udocker) successfully [in Google Colab](https://gist.github.com/mwufi/6718b30761cd109f9aff04c5144eb885?permalink_comment_id=4673355#gistcomment-4673355). `udocker` appears to be a way to interface with Docker images without needing to install the full Docker program:
     > - udocker does not provide all the docker features, and is not intended as a docker replacement. udocker is mainly oriented at providing a run-time environment for containers execution in user space.
     > - udocker is written in Python, it has a minimal set of dependencies so that can be executed in a wide range of Linux systems.
     > - udocker does not make use of docker nor requires its presence.
  - That said, we would have to go back to maintaining a Docker image, which runs into our past concerns from: https://github.com/spinalcordtoolbox/spinalcordtoolbox/issues/2965
  - Meaning, the question becomes: "developer time to set up and maintain the image" vs. "benefits from saving time for end users"
  - Additionally, since what we are doing is "non-kosher" in terms of Google Colab, this solution may be brittle / prone to breaking in the future.
- Using solutions other than Docker:
  - **Storing venv in Google Drive**: It seems as though Google Colab can interface with Google Drive fairly easily. Given this, it seems like some folks have tried to store a virtual environment on Google Drive, then mount the folder to re-use the virtual environment ([1](https://netraneupane.medium.com/how-to-install-libraries-permanently-in-google-colab-fb15a585d8a5), [2](https://stackoverflow.com/a/61926411)).
  - However, I have concerns about the potential performance of running SCT this way, as I imagine the interface between Colab + Drive is (relatively) slow and meant for non-executable files.
  - Plus, this would have all of the same maintenance issues (and then some) as the Docker solution.

tl;dr: It might be possible to do this, but it would depend on whether we consider it worth it to devote further dev time to explore + maintain a solution.

jcohenadad, 2024-01-23:
Thank you SO much for the deep dive into my question @joshuacwnewton ! That pretty much answers it: no need to invest dev time to find a solution that will save a few min of run time (esp. given that the rest of the notebook takes >1h to run because of the many image registrations). I think we can close this then.

closed 2024-01-23 by jcohenadad