shimming-toolbox/rf-shimming-7t #78: Clarify: how should the user supply the sudo password

dpapp86, 2024-01-23:
I suspect @jcohenadad chiefly runs this as a google colab, while I run it as a local notebook. At least in the local notebook, there is no straightforward way to enter text into the cell (or at least none is pointed to by the notebook), so the notebook hangs on this step:

[image]

I have installed the requirements, and I recall doing so via this notebook, but I have no recollection how I did it. I think I just force-ran it a few times.

jcohenadad, 2024-01-23:
When the notebook is run locally, this cell should not be run (as instructed after the ℹ️)

dpapp86, 2024-01-23:
User should RTFM, indeed. Closing.

closed 2024-01-23 by dpapp86