shimming-toolbox/rf-shimming-7t #85: Number and content of figures/tables

dpapp86, 2024-01-25:
Related to #84

I think the figures should be:

1, workflow

2, B1+ maps across the shim conditions for one volunteer (so what is produced by #72 )

3, B1+ value as a curve  (as figure) and stats (as a table) for B1+ along the SC (C3 T2) across volunteers and shim conditions

4, CSF/SC signal intensity ratio curves (as figure) and stats (as table) along the SC across volunteers and shim conditions 

5, MP(2)RAGE as a pretty comparison/signal intensity curve/stats

dpapp86, 2024-01-25:
As per https://github.com/shimming-toolbox/rf-shimming-7t/issues/84#issuecomment-1910256917 , currently at 4

jcohenadad, 2024-01-25:
do you think it will be 'cosmetic' enough to merge curves and tables within the same fig? The formatting is quite different. I was instead thinking of merging the 2 sets of plots into one fig, and the 2 tables into another fig (ie: big table).

dpapp86, 2024-01-25:
So B1+ curves and CSF/SC curves is one fig, and then the tables in another? That sounds OK. I was thinking more procedurally (create X and Y from data, merge into figure), but you are correct, your approach will look nicer

jcohenadad, 2024-02-13:
All resolved-- we can close

closed 2024-02-13 by jcohenadad