shimming-toolbox/rf-shimming-7t #92: Compare manually drawing a rectangular ROI with the results derived from SC segmentation?

dpapp86, 2024-02-13:
From Kyle's comment here:
https://docs.google.com/document/d/1NFWOPu-yP1u0O62YUX4w5IrzwaRwaMozwdJPhJ32XLA/edit?disco=AAABF6huiJ0

To do a proper comparison, I would need to draw an angled rectangle for the mask, and I don't know how to do that using SCT. 

But as a quick and dirty comparison, I did the following: The current pipeline uses a spinal cord segmentation as input for the shimming coefficient calulation in ST, but the ROI is a failry narrow cylinder around it. I merely modified so that we now calculate in a rectangle drawn around the SC segmentation, then adjusted the rectangle to be as wide in A-P and R-L as the B1+ shim box was for Volume Specific shimming. So the ROI still followed the curve of the spine (as opposed to merely being an oblique box), but it was much closer to the one used by the scanner. 

I can only compare the CoV calculation that ST itself does (it combined the measured B1+ maps using the new, calculated shim values, and then takes a CoV within the ROI), which is not a like-for-like with out experiment, but still, for sub 04, using the COV reduction RF shimming approach, I see 22.1% using the spinal cord+cylinder mask, and 27.5% using the rectangular mask. So this might be worth exploring further and putting into an appendix.

Moving discussion here @jcohenadad 

Pls ping me if this is of interest (and if I have not explained it wll, it is late here)

jcohenadad, 2024-02-14:
amazing-- that's exactly what we need. If you can take a screenshot of both masks, we report the numbers, and that's all we need

dpapp86, 2024-02-15:
I've put together the numbers and the two masks (overlaid) in a figure for Sub04 here: https://docs.google.com/presentation/d/1SY4UtXNas9o-S4GYaqGDehS0KoUiwA-ISRyhL6LHxAA/edit#slide=id.g2b9660979b0_1_0