shimming-toolbox/rf-shimming-7t #97: Original submission: Reviewer 2 comments

dpapp86, 2024-03-21:
> Comments to the Author
> The authors recognize that 7T spinal cord imaging poses different challenges than those of the brain and there are necessary changes that need to occur to improve image quality and applicability.  As such one area that is addressed in this manuscript is the challenge of B1 homogeneity throughout the cord.  To address this, the authors compared vendor supplied and improved phase/amplitude RF shimming in the cord at 7T and compared B1 efficiency as well as CSF/cord signal ratio.  Their findings indicate that improvements over vendor supplied methods can be had which may improve the signal fidelity in the cord at UHF.  While there are several novel pieces to this manuscript, I have some notes below that hopefully can be used to improve the manuscript.  Overall it is well written and clear. 
> 
> Comments
> 1.       (minor) in the introduction, first paragraph, the authors discuss higher detection of cortical lesions as a benefit from SC imaging.  While true for brain, not so for the cord

R1.1: But this **is** true for the cord, no? Or are they objecting to the word "cortical"?

> 2.      The labelling of the methods caused some confusion to me on first read.  For example, in the paragraph leading up to Eq 1, the authors state, “…the coefficient of variation (CoV) of the combined B1+ field was minimized within the ROI by adjusting the phase…” in the last paragraph on the same page, the authors state, “to optimize for CoV reduction (referred to as CoV) the CoV of the combined B1+ field was minimized by adjusting both the phase and magnitude.  So, in each case, the goal was minimize the CoV, however, optimizing the CoV reduction (called CoV) caused some confusion when I was reading it.  Perhaps the authors can clarify the language to discriminate between minimizing the CoV and the method whose goal it is to minimize the CoV. 

R2.2: After not having looked at the article for a while, I agree, it is confusing on a re-read. I think we should explicitly change the name of the _shimming method_ "CoV" to "CV_red" (coefficient of variation reduction).

> 3.      When considering Figure 2 and Table 1, it is stated that SAReff is the method that maximizes the B1 efficiency, however, when looking at Figure 2 and Table 1, both the CP and SAReff are effectively the same.  Can the authors comment on the similarity (granted there is a small difference, but likely not significant) between a very basic CP method and the SAReff method and why the “contrast” is actually higher in the CP than the SAReff.  I was surprised to see the similarities

R2.3: I am not sure what the correct response is. Perhaps we should reach out to Kyle?

> 4.      Regarding contrast.  The authors use the CSF-SC ratio measured from the GRE scan as their measure for contrast.  This seems somewhat problematic because it may not reflect improved image quality.  That is, if the CSF-SC ratio is higher, it could be that the CSF signal is higher only, the SC signal is lower, or both.  Several recommendations on this:
> a.      Would the authors consider providing a CNR measurement rather than a ratio measurement, with details about the signal in each compartment (CSF and cord)?  This will help the reader understand the source of the improved contrast.  Or justifiy why the ratio is a better measure.

R2.4a: Measuring CNR is not straightforward for scans like ours (artefacted background, etc). 

> b.      While the figures in the manuscript are very good, there are no figures of the actual GRE images (only B1-type maps).  Regarding improved CSF/Cord contrast, figures of the resulting GRE map from each shimming method would be informative to understand how, visually, the images are improved.  I do think it is important to include resulting anatomical images from each shimming method to visually inspect/interpret

R2.4b: I agree, we could move the current Figure 4 (the one showing the two ROIs comparing ST and volume-selective) to Supplementary, and add in the GRE figure to follow the B1+ figure. This would also allow us to discuss how the changes in B1+ efficiency/homogeneity do not translate linearly into homogeneity for the GRE signal. 

Should we include the MPRAGE results in the same discussion?

> 5.      Figure 3 – For A) please consider putting all panels on the same y-scale. It is hard to interpret the variability across participants when each panels is on a different scale.  For B) similarly, please put on the same scale.  But, figure B also highlights comment 4 regarding what the interpretation of the contrast should be.  Especially considering that Sub 1-3 have increasing “contrast” with lower spinal cord levels, but Sub 4-5 have decreasing “contrast” with level.  Does this mean that in the latter two subjects, the image quality is worse than those in the upper 3 panels?  Difficult to interpret these findings.  (minor) What does “linearly scaled” mean when speaking of the curves?

R 2.5: Not quite sure how to tackle the second part of the question (relating to R2.4)

> 6.      Similarly to comment 4, in the last paragraph of the discussion, the authors state “…we have demonstrated the benefit of RF shimming on anatomical MRI scans.”  I don’t believe this was shown as image quality was not interrogated in this manuscript nor were figures shown to provide evidence that the actual image quality is improved.  Only that there is improvement in the B1 efficiency and potentially some improvement in the ratio of cord/CSF signals.  Please clarify and justify more fully.

R 2.6: I believe that including the GRE scans (perhaps just as a figure, perhaps including signal intensity), and the MPRAGE results, will address this.

jcohenadad, 2024-03-21:
@dpapp86 i suggest we do this in a gdoc (easier to edit/comment)

dpapp86, 2024-03-21:
@jcohenadad https://docs.google.com/document/d/1B8F39z-1o7ui4zJtHA-L0QHKluz-dnVHDxsT_kzZoI0/edit?usp=sharing