shimming-toolbox/rf-shimming-7t #101: CSF/SC contrast figure does not have the same y-scale

dpapp86, 2024-05-21:
Hello!

@agahkarakuzu @mathieuboudreau @jcohenadad 

I do not know when this happened, but in the updated figure 4 in the draft text, the y-scale is correctly set to be the same for all five subjects, same with the B1+ efficiency

[image]

However, as of this PR: #100, this is no longer the case.

[image]

mathieuboudreau, 2024-05-21:
I think I had only set the fixed y-scale for the matplotlib plots (first images you show), but not the plotly images (second set). The reason being the plotly images were not the ones used for the manuscript submission, I don't recall the reason but that was Julien's request in one of these threads.

This was the PR where I merged that feature: https://github.com/shimming-toolbox/rf-shimming-7t/commit/63eba66e3bbac888864ad63e93eb8f9ccba12ec5

So if those lines are not there, we would just need to reintroduce them

mathieuboudreau, 2024-05-21:
ps. to actually view the diff, the [nbdime](https://nbdime.readthedocs.io/en/latest/) tool is best suited for jupyter notebook diffs

dpapp86, 2024-05-24:
Ah.

Sorry to bother you again, @agahkarakuzu. Looking at the notes in the paper, we want CSF/SC ratio to be Figure B, and B1+ ratio to be Figure A. Can you please swap?

dpapp86, 2024-05-26:
@agahkarakuzu gentle ping

Thank you!

agahkarakuzu, 2024-05-26:
@dpapp86 I tore down binderhub deployments for maintenance, and that's the only place I can confirm my changes work.

I'll try to get to this on Monday.

jcohenadad, 2024-05-27:
the A and B should be translated slighty towards the bottom, such that the bottom of each letter aligns with the bottom of the x-label

agahkarakuzu, 2024-05-27:
I've been editing the interactive version btw, is that the one you were referring?

jcohenadad, 2024-05-27:
i’m referring to the static image above my post that will be used for the MRM submission (last missing piece before we can resubmit)

agahkarakuzu, 2024-05-27:
FYI those are screenshots of the plotly figure, I'll push the edits you want and will share a highres png if you'd like to go with those.

mathieuboudreau, 2024-05-27:
> FYI those are screenshots of the plotly figure, I'll push the edits you want and will share a highres png if you'd like to go with those.

I think in the initial submission, it's not the plotly figures that were included in the MRM submission but the matplotlib ones that are above it. That's the reason why I had to reintroduce them: https://github.com/shimming-toolbox/rf-shimming-7t/commit/208b1e80ddfd1f4313b649c67c5d07d711a47ae7

mathieuboudreau, 2024-05-27:
@agahkarakuzu let me know if you'd like me to retake this over - no matter to me!

agahkarakuzu, 2024-05-27:
[image]

@mathieuboudreau I've just pushed the changes @jcohenadad requested to `agah-patch1` branch for which this PR was created. If there's anything else, feel free to take on as I think it would be more efficient! 

[image]

jcohenadad, 2024-05-27:
[image]

placement is perfect, but the fonts should be black (they seem dark grey) to be consistent throughout the manuscript

mathieuboudreau, 2024-05-27:
On it

jcohenadad, 2024-05-27:
@mathieuboudreau since you read this notif-- pls invite me to the ADS meeting-- urgent-- thx

mathieuboudreau, 2024-05-27:
[image]

Here's the latest figure - happy to do more changes if needed

jcohenadad, 2024-05-27:
thank you @mathieuboudreau -- sorry for the nitpick but the A and B should be translated slightly, such that the bottom of each letter aligns with the bottom of the x-label (Vertebral Levels), and the left side of "A" aligns with the top side of the y-label.

jcohenadad, 2024-05-27:
another nitpick: would it be possible to use Arial font (to be consistent throughought the manuscript)

mathieuboudreau, 2024-05-27:
No problem! Brb

mathieuboudreau, 2024-05-27:
@jcohenadad Here's the latest version:

[image]

jcohenadad, 2024-05-27:
amazing, can you please make the legend font slightly bigger?
also, if not complicated, close the box on right and top
thanks!

mathieuboudreau, 2024-05-27:
@jcohenadad Here's the newest version!

[image]

jcohenadad, 2024-05-27:
- the legend is overlaying quite a large part of 'useful' information-- would it be possible to shift it towards the bottom left side of the figure axis (where there is less curves)
- possible to slightly increase the size of the other labels (x-lab, y-lab, and axis lab), maybe use the same size as the legend for consistency
thx!

mathieuboudreau, 2024-05-27:
Here's the update!

[image]

jcohenadad, 2024-05-27:
that's a winner! thank you for your patience

mathieuboudreau, 2024-05-27:
No problem! I'll upload it to the paper's folder now and update the manuscript

mathieuboudreau, 2024-05-27:
Done! @agahkarakuzu I've pushed all my changes to the branch you were working on (agah-patch1), should we merge now?

dpapp86, 2024-05-28:
@agahkarakuzu @mathieuboudreau Please merge ASAP, we need to submit the revised paper ASAP, and we want to include an up to date link for the proper version.

Please ping Julien when you have merged

agahkarakuzu, 2024-05-28:
Oh I see sorry, I thought you were mentioning the version on neurolibre. 

I am lost here a bit too given that I am not actually a part of this, but if what's needed is a release, I can do that.

mathieuboudreau, 2024-05-28:
>I am lost here a bit too given that I am not actually a part of this, but if what's needed is a release, I can do that.

Let me merge and make the release; we need to update the link where the badges point on the README, and provide the version link for the paper. I'll update here when I'm done.

As agah mentionned, the neurolibre version / link can't be updated (it's a preprint, so only for the version prior to the initial submission).

mathieuboudreau, 2024-05-28:
Published the release (https://github.com/shimming-toolbox/rf-shimming-7t/releases/tag/r20240428), now checking that the notebook executes on Colab since Agah and I were developing for the PRs on the NeuroLibre servers, brb

jcohenadad, 2024-05-28:
should be 20240528 no?

mathieuboudreau, 2024-05-28:
Woops, yes you're right I got the month wrong, my bad. Fixing it

mathieuboudreau, 2024-05-28:
Here's the release version: https://github.com/shimming-toolbox/rf-shimming-7t/releases/tag/r20240528

Here's the link to the actual code for that version: https://github.com/shimming-toolbox/rf-shimming-7t/tree/r20240528

Here's the link to Google Colab for that version: https://colab.research.google.com/github/shimming-toolbox/rf-shimming-7t/blob/r20240528/content/index.ipynb

dpapp86, 2024-05-28:
A thousand blessings on your house!

Pinging @jcohenadad

jcohenadad, 2024-05-28:
Fixed in https://github.com/shimming-toolbox/rf-shimming-7t/tree/r20240528

closed 2024-05-28 by jcohenadad