shimming-toolbox/rf-shimming-7t #102: Update GRE figure spacing

dpapp86, 2024-05-27:
@agahkarakuzu 
@mathieuboudreau 

Sorry, guys. Some last-minute changes for spacing.

In the GRE figure field (the one created here #100 ), change the line:
```
#Create a figure with multiple subplots
fig, axes = plt.subplots(2, 4, figsize=(8, 5))
```

TO:

```
#Create a figure with multiple subplots
fig, axes = plt.subplots(2, 4, figsize=(8, 5.3))
```

And change

```
# Delete unneeded first subplot
axes[0].set_axis_off()
plt.tight_layout()
plt.subplots_adjust(wspace=0.1, hspace=0.05, right=0.9) 
plt.show()
```

TO

```
# Delete unneeded first subplot
axes[0].set_axis_off()
plt.tight_layout()
plt.subplots_adjust(wspace=0.1, hspace=0, right=0.9) 
plt.show()
```

Expected output (OBS! the vertical spacing between the rows is bigger than before)

[image]

mathieuboudreau, 2024-05-27:
@dpapp86 this is the output - looks ok to you?

[image]

dpapp86, 2024-05-27:
@mathieuboudreau Yes

Thank you!

mathieuboudreau, 2024-05-27:
Resolved in b234c5b

closed 2024-05-27 by mathieuboudreau