shimming-toolbox/rf-shimming-7t #105: MRM paper vs. neurolibre inconsistency

jcohenadad, 2024-05-28:
looking at neurolibre publication on 20240528_141628 (https://preprint.neurolibre.org/10.55458/neurolibre.00025/) is inconsistent with r1 version of manuscript, notably:

## fig 4 (formely 3) is different between neurolibre and resubmitted manuscript:

neurolibre

[image]

MRM manuscript:

[image]

(notice the swapped B1+ effici and CSF/SC contrast)

# fig 3 missing:

r1 of MRM article introduced a new figure, not visible on neurolibre:

[image]

tagging @agahkarakuzu @mathieuboudreau and also @nstikov because it is part of the 'user experience'

agahkarakuzu, 2024-05-29:
@mathieuboudreau bioarxiv etc do support that, we will support soon as well. I can update both pdf and interactive versions, the pdf on that page can be updated. Old stuff will go /v1 directory.

agahkarakuzu, 2024-05-29:
@jcohenadad we are introducing myst support soon, which will surely improve ux.

jcohenadad, 2024-05-29:
@mathieuboudreau thank you for cross-referencing your comment, indeed you addressed this point specifically

> As agah mentionned, the neurolibre version / link can't be updated (it's a preprint, so only for the version prior to the initial submission).

beyond the fact that places like arXiv indeed make it possible to update the preprint version (as Agah mentioned already), my view/understanding of neurolibre is that it is not a 'traditional' preprint. To me, a preprint is, well, as the name says, a preprint of the submitted manuscript, which is essentially a static duplicate of the submitted draft to a journal. However, neurolibre offers more (ie dynamic objects vs. static), and I see that more as a _companion_ publication (with more, richer information) rather than a duplication. In fact, the title of our neurolibre publication ("Analysis code for the paper "RF shimming in the cervical spinal cord at 7T") differs from the article.

mathieuboudreau, 2024-05-29:
> beyond the fact that places like arXiv indeed make it possible to update the preprint version (as Agah mentioned already), my view/understanding of neurolibre is that it is not a 'traditional' [...]

In principle, I agree with everything you said. In fact, this was an issue I had to address when resubmitting the T1 challenge paper, and that was a pain to handle in regards to NeuroLibre because our resubmission went from a Full Paper to a Technical Note, drastically changing the final paper vs the original NeuroLibre submission. Our solution, which I don't quite like but at the time we were limited in options, was to resubmit another preprint to NeuroLibre:

* First T1 challenge neurolibre preprint: https://neurolibre.org/papers/10.55458/neurolibre.00014
* Second  T1 challenge neurolibre preprint: https://neurolibre.org/papers/10.55458/neurolibre.00023

This is very messy for a lot of reasons, in particular because we have two preprint DOIs for the same work but presented differently (long Full Paper format, and short Technical Note format). But, because Nikola quite strongly wanted a NeuroLibre link that her could share that more closely reflected what would become our final paper, this is the route we all agreed on for this paper.

As Agah mentioned, there seems to be (currently) some technical limitations on the backend of NeuroLibre and how they handle DOIs and links such that it prevents authors to submit revisions, and he's working on that.

One last note however, is that another limitations may be related to copyright. I know from firsthand experience, that some publishers have strict guidelines surrounding "preprints" or just sharing versions, and the final published version. While working on our T1 chapter, we reached out to Elsevier to ensure that they would not ask us to pull down our interactive blog post after the book was published. Their response was that, as long as we openly share only the content that we created prior to submitting it to their portal, that our creative commons license applies and they couldn't touch that. However, they said that we would not be allowed to update our blog post after we've submitted the chapter through their portal and received feedback from the reviewers, as it appears their copyright starts somewhere around there (maybe because some of their ressources are being used during the review process, like handling editors and websites?).  

That being said, I doubt that MRM would care if you updated the content of your neurolibre preprint, however one could see a situation that if your paper became wildly popular and impactful, that Wiley would prefer readers go to their website and pay for their licenses to read published content, instead of a NeuroLibre preprint that was updated to match the final paper exactly.

jcohenadad, 2024-05-29:
Thank you for sharing these insights @mathieuboudreau, I understand better the additional challenges wrt. handling a neurolibre publication.